rkrajnc/minimig-mist #11: User Bug Report

rkrajnc, 2015-05-27:
Submitter: cplnobby
Email: cplnobby@gmail.com
minimig-mist release: minimig_aga_20150518_r1057
firmware version: firmware_150524_joystick_hat
kickstart version: 3.1
chipRAM: 2MB
fastRAM: 4MB
slowRAM: 1.5MB
CPU: 68010
GFX: OCS-A500 NTSC

game / demo: F-19 Stealth Fighter V2.0    (16.11.08, done by JOTD) (F19StealthFighter_v2.0_0117)
WHDload 18.0

other system tested: WinUAE, real A1200 w/68030@40 128MB

description of the problem: Game crashes after playing for a bit, takes at least 10-15 minutes before crash usually. Starting a mission on training and setting autopilot and letting the plane fly itself for some time will usually show it.  More than one mission may be required.

Sometimes screen display is just completely trashed with game crash, sometimes crash to WB ex. (Exception "Line 1010 Emulator" ($20) at $20E80 occurred)

rkrajnc, 2016-02-25:
Is this still a problem with the new minimig_mist-1_2 release?

AmigaIstMiST, 2016-05-11:
I had a similar problem with FA-18 Interceptor. After a few minutes, the screen was freezing, without any error message, so I had to do a reset. But after disabling fastram, the problem disappeared, so I think, this was an old game incompatibilty with fastram. Maybe it's worth a try to retest this with chipram only?