rkrajnc/minimig-mist #9: Minimig core crushes Workbench 3.1 Install Disk

rkrajnc, 2015-05-27:
https://code.google.com/p/mist-board/issues/detail?id=32

What steps will reproduce the problem?
1. Create empty hardfile and mount it
2. Boot MiST from Workbench 3.1 (Install) Disk
3. Partition hard drive with HDToolbox
4. Reboot the virtual Amiga after partitioning
5. Amiga throws checksum error at around block 885 of Workbench Install Disk

What is the expected output? What do you see instead?
Simulated Amiga reboots fine and doesn't throw an error on Workbench Disk 1. I tested the same empty hard file and the same ADFs on the latest WinUAE - no problems at all.

What version of the product are you using? On what operating system?
Latest Firmware and core
Kickstart v3.1 r40.63
Workbench 3.1 (tested with Commodore and ESCOM disk sets)
HDF created by latest WinUAE and a zeroed file

Please provide any additional information below.
I attached the working ADFs and the ADFs after using them in MiST. Maybe someone might find out what went wrong. My SD-Card is working, I checked it using h2testw. Saving on floppy images generally works, I could save and load a game on it's save disk.

rkrajnc, 2015-05-30:
Can't reproduce this bug.

Make sure you are following this steps:

- create a new, empty file and copy it to sd card
- in the OSD settings -> hard disk settings, set master to hardfile (disk img), select your disk image, AND disable the slave (must read 'Disabled')
- apply changes and reboot as instructed by OSD
- insert your workbench install disk
- start HDToolbox
- make sure only ONE device is seen, otherwise you did something wrong
- make sure to use Change Drive Type -> Define New -> Read Configuration and save before continuing
- partition your drive as you wish
- reboot
- install Workbench

If you follow this steps, everything works OK. You probably either used wrong disk geometry (didn't use 'Change Drive Type' and 'Read Configuration') or you tried to install on the whole SD card (which I believe is the default option for slave, which, like I said, should be disabled!)

Also, as is recommended anyway for any Amiga, use fast format option.

Another thing:

I hope you are aware that minimig, like any real Amiga with IDE drives, needs to have MaxTransfer set to 0x1fe00, otherwise you will have problems.

HDToolBox -> Partition Drive -> Advanced Options -> Change -> MaxTranfer and set it to 0x1fe00 for every partition (and confirm by pressing Enter key after entering the number)

rkrajnc, 2015-06-30:
This bug can't be reproduced, closing for now. Will reopen if I'll ever be able to reproduce it.

Video showing no corruption on minimig-mist core:
https://www.youtube.com/watch?v=ThCpeKefQic

closed 2015-06-30 by rkrajnc

AmigaIstMiST, 2017-01-31:
Hi Rok, did receive the SD card? I asked Till to forward it to you. After installing WB 3.1 one of the WB floppy images was corrupted. Maybe it is possible to reproduce this issus now.

rkrajnc, 2017-02-01:
Hi,

I didn't get any SD cards. Is this still a problem for you? Did you try different SD cards, possibly different manufacturers?

AmigaIstMiST, 2017-02-01:
I asked Till around november/december last year to forward a SD card with this issue to you. He sent me an eMai and told me he did so. Maybe the sd card got lost on the way :-(

No it is not a problem for me any more, I have another working sd card now.

However, it is possible to buy a sd card for the MiST now at Lotharek and Dragonbox.