rkrajnc/minimig-mist #18: minimig-mist-v1.0 can't use kickstart version less than 2.xx

Unixware, 2015-06-25:
tested on MIST, the core freezes when I m trying to load a version of kickstart 1.3/1.2 , the 2.0 from A600 works fine anyway.
Settings are as of recommended: CPU = 68000, Turbo=NONE, Chipset=ECS, chipRAM=0.5MB, slowRAM=0.5MB.

rkrajnc, 2015-06-30:
Kickstart 1.3 works fine for me.

Does the OSD still show up if you press F12?
Can you try saving the selected config and kickstart rom as the default config and try rebooting?
How does the freeze look like, are there any corrupt graphics on the screen, or is the screen a specific color?

Unixware, 2015-07-03:
middle led (orange) is blinking and there is no signal (monitor goes to sleep mode). same results when save the setup as 'default' and (re)boot

mkdj69, 2015-07-08:
Hello :)

I too have been having trouble with all AGA versions of the MIST core - non AGA accepts my 1.2/1.3 - but try this:

Kickstart v1.3 r34.5 (1987)(Commodore)(A500-A1000-A2000-CDTV)[o]

It's an overdump ( 512KB ) and now all AGA cores needing 1.3, including 1.0, boot fine.

Markus.

Unixware, 2015-07-08:
cheers for this, works!

rkrajnc, 2015-07-08:
Looks like I forgot to implement support for 256kB ROMs, I'll fix it soon.

So 512kB ROMs work for both of you, it's just the 256kB ones that fail?

mkdj69, 2015-07-08:
Hi Rok,

So far I have only tested:
  
  Kickstart v1.2 r33.180 (1986)(Commodore)(A500-A1000-A2000)[!]
  Kickstart v1.3 r34.5 (1987)(Commodore)(A500-A1000-A2000-CDTV)[!]

which both fail but:
  
  Kickstart v1.2 r33.180 (1986)(Commodore)(A500-A1000-A2000)[o]
  Kickstart v1.3 r34.5 (1987)(Commodore)(A500-A1000-A2000-CDTV)[o]

both work.

So yeah, the 256KB ROMs are not loading ... OSD still works but after time display goes offline and sleeps ( using a LCD TV via VGA ).

Markus.

Unixware, 2015-07-08:
I have exactly the same results with Markus , esp the video signal goes off, when I use 1.2/1.3 256KB ROM images

rkrajnc, 2015-07-08:
OK, thanks for the report guys, I'll try to fix it as soon as possible.

mkdj69, 2015-07-10:
OK, no worries Rok ;)

rkrajnc, 2015-09-09:
Hi, I added support for 256kB kickstart ROMs in the firmware. I'm still working on the AmigaForever encrypted ROMs.

Unixware, 2015-09-09:
cheers!

closed 2015-09-09 by Unixware

rkrajnc, 2015-09-11:
Fixed in minimig-mist-1_1 (52f1216a899418c61dd1a10840b3f0622af46bc5).