rkrajnc/minimig-mist #20: Ishar 1,2,3 and Robinson's Requiem

FenioPL, 2015-06-27:
firmware version: ATH150622 and older.
kickstart version: All. From 1.3 to 3.1
chipRAM: 1-2mb
fastRAM: 0-24mb
slowRAM: 0.0mb
game: Ishar 1 / Ishar 2 / Ishar 3 / OCS or AGA version & Robinson's Requiem AGA. Floppy and WHDLoad version.
description of the problem: mouse pointer not work correctly. https://www.youtube.com/watch?v=qUTcpa_-ceM

rkrajnc, 2015-09-10:
It seems these games also use the JOYTEST register to set mouse counters. I fixed the issue and don't see any strange mouse movements. Will be available in next minimig-mist release.

If you still see problems, feel free to re-open this issue.

closed 2015-09-10 by rkrajnc

rkrajnc, 2015-09-11:
Fixed in minimig-mist-1_1 (52f1216a899418c61dd1a10840b3f0622af46bc5).