rkrajnc/minimig-mist #24: Issues with X-Out - Core crash

Ricky2u, 2015-09-24:
minimig-mist release: AGA 1.1
firmware version: latest release from github website as of today
kickstart versions: 1.2, 1.3, 2.04
chipRAM: 512kb
fastRAM: 0
slowRAM: 0 & 512kb
game / demo: X-Out (1990)(Rainbow Arts)(Disk 1 of 2)[cr PNA][t +2 PNA]
other system tested: WinUAE - works correctly in WinUAE with Kick 1.2 and 512kb CHIP only

description of the problem: When loading the game after the group scene demo the yellow core light flashes constantly and then the MIST needs to be hard reset to work correctly.

Kick 1.2 + 512KB Chip only: After the scene demo and you click left mouse button to continue the guru errors comes up on screen and the whole MIST freezes and comes up with a minimig OSD error: Error: WriteTrack Code: #27

With kick 1.3 and 2.04 with 512 or 512+512 The core light keeps flashing and may just reset or the MIST becomes unresponsive and the core needs to be hard reset. One time it even made a long tone noise with a blank screen.

Tried different copies of the same game and all replicate the same bug.

Thanks,
Ricky2u

rkrajnc, 2015-11-26:
Hi,

this is probably a CPU core problem, I'll see if it can be fixed, but it is going to take some time unfortunately.

rkrajnc, 2016-02-25:
Is this still a problem with the new minimig_mist-1_2 release?

Ricky2u, 2016-02-25:
Hi,

I just tested the latest Minimig AGA 1.2 core with firmware dated ATH160123

Unfortunately I can replicate the same issue and I get in A500 config mode this guru error after the cracktro: #0000000A.00C05128

When in A1200 config mode I get a red screen loop and the core needs to be hard rebooted.

Cheers.