rkrajnc/minimig-mist #25: aibb crash

juri74, 2015-11-07:
Hello i use minimig aga v1.1 and latest mist firmware, i'm not able to make a module with aibb. With only 4mb of fast ram i was able to make a module but strange things happens:
The lower half of beachball was really glitched, with 4mb showconfig report 4mb of 16bit ram but aibb report presence of 32bit ram..
Aibb always hang randomly in one of the floating point test (except beachball)

juri74, 2015-11-07:
Forgot to write that aibb hang with presence of 32bit memory (24 mb selected in settings)

rkrajnc, 2015-11-08:
Did you try disabling turbo for chipRAM and/or Kickstart in OSD menu?

juri74, 2015-11-10:
ok so i did further testing, unfortunately i completely forgot that i was using the experimental 020 cpu emulation (forgive me i'm getting old) with 68000 cpu emulation all run fine :) however if it could help to improve 68020 cpu emulation (or minimig aga core) here some test i made with aibb (all with 68020 core)
020+0mb ram + turbo (any setting) result OK
020+4mb ram + turbo (any setting) result OK
020+24mb ram + NO TURBO result OK
020+24mb ram + CHIP TURBO ON + KICK TURBO OFF result OK
020+24mb ram + CHIP TURBO OFF + KICK TURBO ON result AMIGA HANG
020+24mb ram + ALL TURBO ON result AMIGA HANG
the curious thing is that AIBB do not craash when executing 020 code but when executing standard (68000) code

regarding SLOW RAM i had problem setting it to MORE than 0.5mb ram... since i usually do not use it, i do not know if on an AGA machine with KICK 3.1 is it possible to have more than 0.5 meg of chip ram, every setting i select (0.5, 1.0 or 1.5) still result of 0.5 meg with showconfig on wb.. the minimig startup screen report the right amount of slow ram selected.

thanks
Juri

rkrajnc, 2015-11-10:
This could be a timing problem of the CPU core. I'm trying to fix the issues the TG68 CPU core has, but the way the core is written ... don't hold your breath ;) There are apparently also some instructions that are not implemented correctly, Till is looking into that.

Regarding slowRAM, yes, I forgot to mention that in the readme - the amount of slowRAM is dependent on the amount of chip memory used - if you use 2MB of chip memory, you can only use 0.5MB of slowRAM. Since no software / game I know of uses more than 0.5MB of slowRAM, I don't think this is a problem. Besides, it is not recommended to use slowRAM when running minimig in AGA mode.

juri74, 2015-11-10:
thank you, no problem for me since i usually do not use slow ram, just turned on to make some deep test with aibb.
I think amiga reimplementation is very good actually, you're doing great work. thanks again

rkrajnc, 2016-02-25:
Is this still a problem with the new minimig_mist-1_2 release?

juri74, 2016-02-25:
hello i tested it (but using last minute core fix for turrican 2 glitches) using turbo=both it worked flawlessly :+1: 
i think it is fixed :)

rkrajnc, 2016-02-25:
Great, thanks!

closed 2016-02-25 by rkrajnc