rkrajnc/minimig-mist #26: HELP key missing?

juri74, 2015-11-21:
i do not know if this is (reported) bug or not but i wasn't able to find any informations.

•firmware version: Latest
•kickstart version: any
•chipRAM: any
•fastRAM: any
•game :Damocles
•other system tested: WinUAE: working, HELP KEY is mapped on PAGDOWN key.
•description of the problem: 
Damocles require pressing HELP KEY to access load/save menu, but after pressing ALL keys on keyboard nothing happended, double checked with KME (http://aminet.net/package/util/misc/kme)
and it appear that HELP key is never pressed.. if it have a keycode i could remap to some other keys (like winuae i would use PAGDOWN key) but i do not know the keycode.

thanks
Juri

rkrajnc, 2015-11-24:
Hi,

honestly, I never used the help key until now ;) It's possible that the mapping is missing, I'll check it out and add a map Help -> PAGEDOWN.

rkrajnc, 2015-11-24:
Can you try the firmware attached (rename it to firmware.upg), the HELP key should be mapped to PAGEDOWN.

[image]

juri74, 2015-11-24:
hello, unfortunately it doesn't work... :'(
now pressing HELP key HRTMon is activated, F11 do not activate it anymore.. i tried to disable it and i noticed that the HRTmon option already reported as deactivated!!! i renamed  it but when i press PAGDOWN amiga crash with guru code 8000 000B 0000 8922
i have an italian keyboard but i don't think this is the problem since all other keys have worked before without problem..

rkrajnc, 2015-11-24:
Ok, it's not a simple fix then ;) I forgot that NMI (F11) handling is done in the FPGA core, so an FPGA update will be needed. I'll let you know when I'll fix it.

juri74, 2015-11-24:
thank you very much! :)

rkrajnc, 2016-02-25:
Fixed in minimig_mist-1_2 (https://github.com/rkrajnc/minimig-mist/commit/da69f6ec45f9956f1df56f0a9a7f4a062c9b726e).

closed 2016-02-25 by rkrajnc