rkrajnc/minimig-mist #27: Help needed getting off the ground...

dylanmc, 2015-11-25:
Many folks here are running happily, so I have no doubt it's possible, but I'm a bit stuck. I could be thick, or there could be a subtle step (or dependency) either missing or implied.

I have a brand-new (as of Nov 2015 MIST),
my SDCard is FAT32'd, and has the following files:
 * core.rbf - from minimig1.1
 * kick.rom - version 3.1, from my Amiga Forever purchase, transferred from my Android phone
 * HRTMON.ROM, from minimig1.1
 * minimig.art, minimig.bal, minimig.cop also from minimig1.1
 * ... and a directory full of the Workbench diskettes.

When I boot, I get the spinning ball miniMIG prompt, with the configuration identified:
```
CPU: 68020
Chipset: AGA
CHIP: 2.0 MB, FAST: 24.0 MB, SLOW: none
```
I've tried attaching workbench diskettes, games, etc. to df0:, but when I exit the popup menu, the screen stays black, and nothing else happens.

I've been able to get NES and Mac Classic going, so have had some success, but feel pretty clueless here. Any help (and fixes to the documentation if needed) would be welcome - thanks!

rkrajnc, 2015-11-25:
Hi,

if you see the spinning ball that means the SD card is working and MiST is reading files from it fine. What do you see on your display after the intro screen with the spinning ball? Normally, if everything is working OK, you should see the familiar Amiga kickstart screen with the floppy disk.

If you don't see it, that means that either your version of the kickstart is not working, or there was something wrong with uploading it to minimig memory.

How did you transfer your kickstart image from your Android phone? Did you use transrom program? If you didn't and you just copied the files from Android, did you also place the rom.key file on the root of the SD card? Since the AmigaForever ROMs are encrypted, you will need the rom.key file to make them work. Please check the size of your kick.rom file in bytes (not kilobytes or MB!) and paste it here.

Can you try putting the aros.rom file from the minimig zip file to the root of the SD card, and renaming it to kick.rom. Also delete all minimig*.cfg files that are in the root of the SD card.

It could also be a problem with the AmigaForever kickstart decryption - I honestly didn't test all of the AmigaForever ROMs, but at least two versions worked OK for me.

Oh, you did update the ARM firmware on your MiST board? Can you paste the version number (open OSD menu, go to the rightmost menu, there you should see the ARM firmware version).

If not, put the firmware.upg from the minimig zip file on the root of your SD card and update the ARM firmware through the OSD menu.

rkrajnc, 2015-11-25:
Ah, I see you didn't update your ARM firmware. You should upgrade with the firmware.upg included in the minimig-mist-v1.1 zip file. Place it on the root of your SD card, start a core (minimig or atariST), open the OSD menu (F12), and find the option to update the firmware - on minimig, the option is the rightmost menu (press right a few times in the OSD menu).

Alternatively, you can use one of the newer firmwares from here:
https://github.com/mist-devel/mist-binaries/tree/master/firmware
download the firmware_XXX.upg file (click on it, then click raw, or use this direct link: https://github.com/mist-devel/mist-binaries/raw/master/firmware/firmware_151118.upg). Rename the file to firmware.upg, place it on the root of your SD card and upgrade the firmware throught the OSD menu like described above.

dylanmc, 2015-11-25:
Yup - updating the firmware was the trick. This was surprising, given I got the MIST this week, and the instructions say "usually, there is no need to upgrade the firmware, unless it is specifically mentioned to do so". It would be helpful to make it more obvious when firmware upgrades are needed. I had the mistaken understanding my firmware was likely newer than the one in the minimig release directory.

closed 2015-11-25 by dylanmc

rkrajnc, 2015-11-26:
I think MiST is always shipped with the same firmware, it was only recently updated to a newer firmware. That is understandable, so the baseline config of all MiST boards is the same, and you can more easily know what kind of problems people might have.

The minimig release readme file mentions that you should upgrade your firmware if it is older than the core release, so it *is* specifically mentioned to do so ;)

Maybe I should add a note in the readme for how to check the firmware version. I think all cores have a readme which explains when the core requires an upgrade to the firmware. I didn't know what kind of firmware is on the new MiST boards, since I have nothing to do with making / selling of MiST boards.

eddygarcas, 2016-07-06:
Thanks both of you for this post, I'd got the very same issue, now fixed :+1: