rkrajnc/minimig-mist #28: scandoubler_disable=1 not applied before kickstart.

sorgelig, 2015-11-25:
Hi,
It looks like scandoubler_disable=1 option from mist.ini is not applied till kickstart loaded.
So, banner with minimig animation and core info looks unsync'ed if i connect to scaler with RGBS-only support.

rkrajnc, 2015-11-27:
I already tried fixing this, but for some reason the scandoubler setting was ignored. I'll take another stab at this soon.

sorgelig, 2015-11-29:
Did you think about implementing YPbPr with Sync-On-Y? I think, it shouldn't be a big problem, just small formula to convert color space. 
Can you implement it as a test feature with key in ini file? 
I think it's step to right direction because component video has standard up to 1080@60p unlike SCART-RGB which is only 576i@50.  All interlaced modes can be internally converted to progressive component video. Also, it's hard to find devices with SCART/RGB support outside of Europe.

I've used to program in almost any language in last 30 years, but i didn't touch FPGA yet :)

rkrajnc, 2017-01-16:
Fixed.

closed 2017-01-16 by rkrajnc