rkrajnc/minimig-mist #31: GM 810000FF with Classic Workbench Full v28

ArKay74, 2016-01-10:
Getting an 810000FF after the Workbench screen is opened. 

810000FF : A quick interrupt has happened to an uninitialised vector

I can supply a compressed HDF if needed.

rkrajnc, 2016-02-25:
Did you try the new release, do the problems still persist?

ArKay74, 2016-02-25:
Of course I have formatted the SD card since trying the installation. Is 
this supposed to be fixed then?
I could give it another try when I find the time for it.

(#31)

ArKay74, 2016-02-26:
Well, this time it worked. That could probably also be due to the fact that I didn't activate the AGA Copper  background effect. Could that have been it?

rkrajnc, 2016-02-26:
The copper background shouldn't be a problem, but I think you can still enable it in the menu (the option should be Copper). Can you try enabling it and report if there are any problems?

ArKay74, 2016-02-26:
After the previous installation the harddisk image didn't even work in WinUAE so I have no idea what went wrong (there were only some weird bars on the screen). Anyway, I did a re-installation with AGA coppers activated and so far it has been stable. It's probably safe to close this bugreport.

closed 2016-02-26 by ArKay74