rkrajnc/minimig-mist #34: Graphic glitches

juri74, 2016-02-24:
hello,
i have a bug report, config:
latest build (1.2)
kick3.1 (and kick1.3)
24mb fast, 2mb chip
ecs/aga 
68020/68000
game: Turrican 2 (whdload, registered user)

game start normally title screen appear to be right, 
but when you begin to play screen fill with small lines

FenioPL, 2016-02-24:
Graphics error in Turrican 2 occur only in AGA chipset enable. On OCS or ECS game works fine. The same problem on Battle Squadron, R-Type II and more games writen on OCS/ECS. On older core working fine with AGA chipset and disabled turbo for chipram but on last core there are errors in graphics if the chipset is set to AGA, even is turbo chipram is disable.

rkrajnc, 2016-02-25:
Try this bugfix release with slower CPU-chipRAM speed when TURBO=off:
http://meditation.somuch.guru/minimig/files/mist/dev/minimig_mist-20160224.zip

(note, this has slower CPU-chipRAM speed than what is expected on an AGA Amiga when turbo is off, so you will probably have to toggle TURBO on & off for certain games)

juri74, 2016-02-25:
ok, little update:
i forgot to reset when switching from AGA to ECS, with ECS turrican works right however i have a "display" problem on workbench screen: entire screen is ROLLED by 1 pixel to the right, rightmost 1 pixel culumn is displayed at first leftmost culumn of pixels (i can provide a picture if needed)
Bugfix release:
this solve graphics glitches in Turrican II with AGA Chipset, TURBO=off (obviously glitches still remain with turbo=on)
Thank you rkrajnc for the quick "fix" :)
one question how the turbo off/on affect workbench operations? (mainly file copy/move operations)
thanks again for your absolutely great work!

rkrajnc, 2016-02-25:
OK, a better fix for this will be developed in the future.

I'll close this issue for now.

> one question how the turbo off/on affect workbench operations?
With turbo enabled, Workbench use should be generally faster.

closed 2016-02-25 by rkrajnc