rkrajnc/minimig-mist #38: Lo-res screen rolled to the left

juri74, 2016-03-06:
Hello!
configuration: mist with lastest firmware/core (1.2 fix). 68020, AGA, 2MB CHIP, 24MB FAST, KICK 3.1, TURBO=none >>> 15 KHZ output <<<

how to clearly see this bug in action:

Execute Deluxepaint and open a 320x256 at any colours deep. (i tested 32 and 256 colors)
Draw a vertical line at the far left of the screen, you will see the same line appear to the far right of the screen, the line you draw an the left looks very thin (like half pixel wide)

the interesting thing is that there is something wrong with OCS/ECS chipset too, but this time it happen on HIRES screens and in the opposite way:
to see this one in action, boot to workbench and open any window. then move the window to the rightmost possible and you will see last pixel column of the window displayed on the far left of the screen.
i can provide pictures if you want
Juri

rkrajnc, 2016-03-07:
Hi,

yes, pictures would help, thanks!

juri74, 2016-03-10:
hi, here are the pictures
AGA low res bug -> https://goo.gl/photos/5P6FAAFBSvmaSQxs8
OCS/ECS hi res bug -> https://goo.gl/photos/6h7yd8TFLqKzoG4W9
thanks

rkrajnc, 2016-03-10:
Ah, that again :) I was afraid this was some completely new problem ;)

This is the cause of the bitplane shifter, which was upgraded with shres scrolling for AGA. It seems the default delays I set up are not correct - the default AGA screen is moved to far to the right, and vice versa for the ECS. It seems the extra shift will have to be bypassed in ECS mode, and the AGA moved to its proper location. The scroller resolution (and thus offset) is also dependent on the bitplane resolution, so it is possible this problem could only be present for certain resolution / chipset combinations.

I'll fix this as soon as I have some time. Thanks for the photos!

AmigaIstMiST, 2016-04-10:
Maybe related (or duplicate) to
https://github.com/mist-devel/mist-binaries/issues/9

rkrajnc, 2016-04-14:
Hi @AmigaIstMiST,

the dot on the screen is most probably a separate issue. I didn't see this myself yet, but I'll check if I can reproduce it.

AmigaIstMiST, 2016-04-14:
Hi @rkrajnc ,

I am not a HW developer, but a question comes in my mind:
your are combining AGA and OCS/ECS in _one_ core, but weren't the A1200 and the the A500/600 two different graphic architectures? Why not two separate cores, one for A1200 and one for A500/600, wouldn't this be easier?

rkrajnc, 2016-04-14:
Yes, AGA and ECS were different architectures, but AGA is heavily (mostly) based on ECS, and it is also very compatible with ECS.

Since the current design already has support for both, it would be (tedious) work to split ECS from AGA. It would also mean maintaining two code bases, which means fixing common bugs in both of them, which is even more tedious, boring work. As far as I'm concerned, I'll keep it like it is at least until a new (hypothetical) CPU core is made with 32 bit external bus (which would then be used exclusively for the AGA core).

rkrajnc, 2016-04-15:
Yes, the real 68020 has a 32bit bus, but the TG68 CPU core used in minimig does not, which requires some hacks to make it work like in a real AGA Amiga, unfortunately with some side effects that haven't been fixed fully yet.

sblendorio, 2016-08-01:
Hi, I confirm that strange behavior in Lo-Res mode.

sblendorio, 2016-08-22:
The same can be notice playing Monkey Island

juri74, 2016-12-21:
hello  @rkrajnc ,
i noticed google removed my pictures for some reason, please tell me if you still need them,
i will take new ones and try to use different web space to hold them on
thanks

rkrajnc, 2016-12-21:
Hi,

I've seen this problem myself, so I know what to look for :) But if you want, feel free to update the pictures, so that other people can see them and know that this is an already known problem.

jmorgon, 2017-01-14:
an example in AmigaDOS window

[image]

kowoba, 2017-01-17:
Here is what I see in Worms DC - I do not know if this is the same bug or not, but thought I'd just post it :) Some columns from the "far left" show up on the "far right".

[image]
[image]

sblendorio, 2017-01-17:
The usecase of AmigaDOS window lacking right border always happens in
ECS/OCS

AmigaIstMiST, 2017-02-21:
Hi @rkrajnc ,

just an idea: is it be possible to do a "ECS only" build without this bug, until you have found the problem?
Or is it more complicated? Most games+programs are OCS/ECS only, anyway.

uchristo, 2017-06-08:
Any further thoughts on that issue?  Can you point out, where to change the delay so I could compile the core by myself to try out....

rkrajnc, 2017-06-09:
The bitplane shifter is here: https://github.com/rkrajnc/minimig-mist/blob/master/rtl/minimig/denise_bitplane_shifter.v

The problem is *how* to change the delay, so it will work OK for both AGA and ECS modes.

uchristo, 2017-06-09:
So..... the bitplane Shifter should be able to read out the value of module 
https://github.com/rkrajnc/minimig-mist/blob/master/rtl/minimig/minimig.v
Line 382: wire	[4:0] chipset_config; //chipset features selection
?

AmigaIstMiST, 2017-08-13:
[image]

This is the AGA Version of Simon The Sorcerer.

uchristo, 2017-10-13:
as I got my OSSC yesterday, motivation has returned as well.
I had a look into the Code again ( has been approx 10 years since I last coded [more or less successfully] in VHDL...). As I understand, Denise is instantiated in https://github.com/rkrajnc/minimig-mist/blob/master/rtl/minimig/minimig.v, Line 659 and following. Config (AGA/ECS) is written in Line 678(for ECS, highest 1 Bit set of 5 Bit Register) or 679(for AGA, uper 2 Bits from 5 Bit Config-Register ).
module denise_bitplane_shifter (...) should be able to read these values  (.clk, .clk7_en and so on are used in https://github.com/rkrajnc/minimig-mist/blob/master/rtl/minimig/denise_bitplane_shifter.v starting in line 29 as well).
Am I correct?

rkrajnc, 2017-11-03:
No (good) solution yet, sorry.

uchristo, 2018-01-03:
Whilst there's no good solution, does a quick-and-dirty solution exist?

kowoba, 2018-01-03:
Yeah, a temporary fix that only works with AGA is much better than no fix at all :)

uchristo, 2018-05-17:
Hm.... any thoughts on https://github.com/rkrajnc/minimig-mist/issues/38#issuecomment-336374691 ? I'd really like to help

AmigaIstMiST, 2018-05-30:
Fixed in V 1.2.3. I think, this bug could be closed.

kowoba, 2018-05-30:
1.2.3?? Wow, what a wonderful day this is! 😃

uchristo, 2018-06-05:
Seems fixed. Had a quick look at SimonTheSorcererAGA yesterday. Thank you very much!

juri74, 2018-07-23:
yes i tested and look fixed, closing

closed 2018-07-23 by juri74

kowoba, 2018-09-24:
In what commit was this fixed? Or does it only exist in the 1.2.3 beta release? Where are the sources?
As far as I can tell, the fix has not been committed, and hence this issue should not have been closed.