rkrajnc/minimig-mist #39: Version info?

kowoba, 2016-03-07:
Maybe I am missing something, but I cannot find any way to tell from within the core, what version of the core that is running - a simple version string somewhere (f.ex 1.2) and build number (if there is such a thing) would be nice.

rkrajnc, 2016-03-07:
The core version is currently only displayed at startup, there's no way to read the version from the running system. Maybe I could add some way to read it from within the Amiga environment, like attaching to the Autoconf device serial number or something. I'll think about it.

kowoba, 2016-03-09:
I did not mean from AmigaOS (though that would be neat!), but when you press F12. For example, the horisontal scroller that says "Welcome to Minimig!" could be followed by a version string? I actually rarely see the startup info and animation, since at that time, my monitor is still busy finding the signal sync :)

rkrajnc, 2016-04-14:
@kowoba I think I can add that info, will let you know when it's done.