rkrajnc/minimig-mist #41: Instability after switching between Kickstart 1.3 and Kickstart 3.1

Paradroyd, 2016-04-05:
I'm experiencing reproducible instability in my MIST when booting in a basic Kickstart 1.3 configuration, then switching to a Kickstart 3.1/AGA configuration. This happens with the latest 1.2 core as well as the 1.1 core, and also with the latest firmware and at least a couple of previous versions.

I can boot into Kickstart 1.3 and use it with no obvious issues, but after I select a 3.1/AGA configuration from the F12 menu and reboot, as soon as I type a character or two on the keyboard, the MIST freezes, then silently reboots with no errors. After this, until I do a full power cycle, typing anything on the keyboard will cause it to freeze and silently reboot again.

If I set the MIST up to boot straight into the same 3.1 configuration without booting 1.3 first, it seems completely stable.  I have a much more verbose account of this with a lot more information that I left <a href="http://www.atari-forum.com/viewtopic.php?f=101&t=29664">in the Atari forums</a> a few days ago.

Thanks.

Paradroyd, 2016-04-06:
It might be worth noting that "Instability" probably isn't the right word, as once you see what's happening, it's pretty consistent and predictable. 

Since posting this, I talked someone else through reproducing it on their MIST and confirmed that they were seeing the same behavior that I was.  I've also confirmed that the issue exists in version 1.1 of the core as well.

rkrajnc, 2016-04-14:
Hi,

thanks for the bug report, nice find!

I noticed that I sometimes have to press F12 multiple times so it is registered, I thought it was just my keyboard acting up. Looks like some keyboard state or keys get latched somewhere, either in the ARM firmware or the FPGA core. I'll check it out.

Paradroyd, 2016-04-14:
No problem. This one has been around for a long time, apparently. Like you I've been seeing what seemed like MIST intermittently not responding to F12. I didn't figure out pretty recently what was really happening.

For what it's worth, after a configuration change mine always throws away (or more like, does something else) with the first two F12 keypresses, or MIST hardware rightmost button presses. It always responds on the third press. Doesn't seem to be related to specific keyboard hardware either , as I've used several. (Which I guess should be obvious since it affects the 3rd front panel button too).

Thanks for looking into it (and for all the work you do on this)!