rkrajnc/minimig-mist #48: Newtek demo incompatibility

kowoba, 2016-07-09:
I don't know if you have seen this over on atari-forum, so I will just post a link here :)

http://www.atari-forum.com/viewtopic.php?f=101&t=27110&sid=7a2194b4311bbdc554b6a4dcb986a67a&start=500#p297340

Have a great summer!

rkrajnc, 2016-07-09:
Hi,

Thanks for the report! I'll take a look at it when I'll have some time. 

Is it known what kind of mode this is using? If this is some sliced HAM mode, I would expect it should work. Anyway, I'll check it out.

EricGus, 2016-07-09:
I posted some details on this software enhanced display mode in the linked atari forums thread .. I believe it uses copper tricks and rotates the display pallet for each hoziontal scan line.. so I'm thinking the issue has to do with timing..