rkrajnc/minimig-mist #53: Minimig-AGA v1.2: Bar / stripes pattern in dark backgrounds

mambrino, 2016-07-26:
Hi,

I found an annoying problem with minimig-AGA core v1.2. With scandoubler activated in dark consistent color backgrounds a kind of vertical bars/stripes pattern appears. I've tried this with Minimig board and doesn't occur.

Here is a picture in which you can see the effect (Demo: Enjoy The Silents / WHDLoad & ADF):

[image]

And this is with minimig board (no problem):

[image]

Tested with different cables / monitors, and it's not a problem related to this.

Thanks ;)

rkrajnc, 2016-07-26:
Are you sure this isn't caused by your monitor? Can you try on another monitor, possibly a CRT? Alternatively, you could try changing your monitor clock phase setting, that usually fixes it.

mambrino, 2016-07-26:
Tested with LG M1921A, M1721A and Panasonic TX-L32ET5 (only works with scandoubler disabled). After testing, I've found that disabling scandoubler the pattern disappears. Maybe is related to this.

rkrajnc, 2016-07-26:
Can you try changing your monitor's clock phase setting, and see if the pattern moves? If it does, it is caused by your monitor, and you can try to eliminate it by adjusting with the clock offset.

If it does not, and it shows even with dithering disabled, then it is a completely new bug, but one I have never seen myself.

mambrino, 2016-07-26:
Tested with a CRT monitor and no pattern at all. Probably is a problem of these two monitors.

Thanks ;)

rkrajnc, 2016-07-26:
It is possible that the original minimig and the MiST board don't have exactly the same frequency down to the last Hz, that is why it works on one board and not another - even a very small difference could cause this.

closed 2016-07-26 by rkrajnc