rkrajnc/minimig-mist #55: Amiga core and MIDI output

sblendorio, 2016-08-19:
Hi,

I tried to use DeluxeMusic Construction Set using the MIDI ports of MIST, but every time I try to make it play through MIDI, I get a Guru Meditation.

Should I install some driver or set some system preference? Is there some game/demo I could use to test MIDI ports?

Thanks in advance.

sblendorio, 2016-08-20:
Hi, now I have a configuration that doesn't "GURUs", but still no sound is emitted from MIDI keyboard (while using ATARI core it's a simple "plug and play" literally). The software is "Deluxe Music 2", and this is the configuration of MIDI/Serial port I used.

Did you see something wrong? Do you know some DEMO that I can use to test MIDI?

[image]

[image]

[image]

kowoba, 2016-08-20:
Doesn't this need camd.library installed?

kowoba, 2016-08-20:
The "old" library package:
http://aminet.net/package/mus/edit/camd

The "new" (and open sourced) library package:
http://aminet.net/package/mus/midi/camd40

sblendorio, 2016-08-21:
The "old" library package:
http://aminet.net/package/mus/edit/camd

...worked GREAT! THANK YOU!

sblendorio, 2016-08-21:
The "old" library package:
http://aminet.net/package/mus/edit/camd

...worked GREAT! THANK YOU!

[image]

kowoba, 2016-08-21:
Awesome! 👍 Glad I could help :)

closed 2016-08-21 by sblendorio