rkrajnc/minimig-mist #62: Floppy disk corruption on some SD cards long term solution

AmigaIstMiST, 2016-09-19:
It's a known issue that minimig/MiST has problems with corrupt ADF files when using larger SD cards. I had the same problem and solved it by moving to a smaller SD card with 2 GB capacity (thanks to a hint by chaos :-) ).
I don't know why this problem exists when using some SD cards, but I think there should be a 100% solution that works with all SD cards. The current situation is really frustrating especially to new users. On some forum entries, users report that they had to try (=buy) several SD cards, before they found one that was reliable. What about this:

- when an ADF image is inserted, copy the image to a dedicated part of the MiST RAM
- all read/write operations use the RAM image
- when removing the ADF image, the modified copy from RAM is synced to the SD card in a safe way, before the ADF image is ejected

This is just what comes to my mind, when I think about this problem. Maybe other/simpler solutions exist for this problem.

rkrajnc, 2016-09-19:
Ho,

First of all, thanks for all your bug reports,  I appreciate it.

This is probably not a size issue,  ss I am successfully using a 16GB SD card.  It seems some cards are just different then others,  but what exactly is the difference,  I don't know, as I have a bunch of them and they all work,  from 32MB to 16GB.

It could be a ARM -> SD csrd timing issue,  I'll look into that first.

AmigaIstMiST, 2016-09-20:
I had this problem with a SANDISK 8Gb SD card.
2GB SONY works fine.

rootfather, 2016-09-20:
I had this issue with some cards, an 8GB SanDisk and a 32 Transcend.

AmigaIstMiST, 2016-09-20:
Hi chaos,

do you think this is an Amiga only problem? Or is this a general MiST problem, that affects the AtariST, too? If yes, maybe the MiST repo would be a better place to report this.

AmigaIstMiST, 2016-12-11:
Hi @rootfather ,
Lotharek and dragonbox sell SD cards especially for the MiST, maybe you should buy one of these.
Lotharek claims on his site, that the main problem are faked sd cards of bad quality.

http://lotharek.pl/products.php?kid=29
https://www.dragonbox.de/de/36-sd-karten

AmigaIstMiST, 2018-06-12:
Nearly all read errors I had on ADFs went away with V1.2.3! I assume, this bug is fixed now. MinimigAGA is becoming perfect after all these years :-)

closed 2018-06-12 by AmigaIstMiST