rkrajnc/minimig-mist #63: Wrong refresh counter value in SDRAM controller

sonycman, 2016-11-05:
Sdram controller sdram_ctrl.v in Mist v1.2 release uses nine bits width memory refresh counter named refreshcnt.
This counter decreases from 511 to zero once in every 16th tact of 114 MHz clock in phase 1 of sdram state machine.

Thus we have one refresh cycle for approx. 72 microseconds, while SDRAM chip need it to be once for 7.8 microsecond.
We have a potential problem here, if I`am get it right.

rkrajnc, 2016-11-06:
Hi,

thanks for the report. The SDRAM chip used requires 64ms refresh cycles (where did you get the info about 7.8ms refresh?) While the refresh cycle is a little longer than required, this shouldn't be a problem, as in my experience SDRAM usually runs just fine with slightly out-of-spec refresh timings.

Nevertheless, I'll try to decrease the counter value a little bit, but I doubt it will help with the general stability problems.

sonycman, 2016-11-06:
Yes, but memory needs 8192 refresh cycles every 64 ms, does it not?
Thus we have one cycle every 7.8 microsecond.
And not one cycle in 72 microsecond - as in your code.

rkrajnc, 2016-11-06:
It seems you are correct. There should be a autorefresh command for each row.

This is actually not my code, I think it was written by tobiflexx, and subsequently used in all minimig variants that have SDRAM, so it looks like all of them have a potential problem :S

Will definitely look into this, thanks again for reporting it.

sonycman, 2016-11-06:
It seems that author is simply forget to correctly initialize the counter, as it is initial value is all the ones.
But your DE1 minimig design has correctly working sdram controller (different version, maybe) with all the IDLE cycles set to auto refresh. 

You are welcome, Rok.

rkrajnc, 2016-11-06:
I thought idle cycles were used for refresh ;) I should really look into the sdram controller code ...

rkrajnc, 2017-01-01:
Fixed in a464a52f6d8d1055a468f70a2ebd12e4a6b994fa.

closed 2017-01-01 by rkrajnc