rkrajnc/minimig-mist #64: Hardwired demo doesn't work.

sirdavid2, 2016-11-11:
I tried to run Hardwired demo on MIST, but it always says it doesn't like axel (accelerator). I checked the CPU speed with sysinfo and it seems to be a bit too fast (1.15 times faster than A600). Different memory or chipset settings and different core versions don't change anything. Any chance to fix it?

rkrajnc, 2016-11-28:
Hi, thanks for the report, I'll take a look and see if anything can be done.

jmorgon, 2016-12-18:
From .adf doesn't work but running from WHDLoad yes. There are only some color glitches in bathroom animation part.

sirdavid2, 2016-12-19:
I have to check it. But still it doesn't fix the speed problem. We should strive for maximum compatibility with the original Amiga machines.

rkrajnc, 2016-12-20:
https://github.com/sirdavid2 : I agree, but 100% compatibility will be hard to reach unless someone writes a cycle-accurate 68000/68020+ core.