rkrajnc/minimig-mist #65: CD32 PAD Option

SheldonRetrogaming, 2016-11-27:
I'm testing CD32 WHDLOAD Games with a CD32PAD Like DB9 with CD32PAD option enabled but
it does'nt work. work only in game with standard 2buttons option when possible.
same issue with a usb pad.

Firmware: 160910
Core: AGA 1.2
the games tested:
Arabian Nights CD32
Base Jumper CD32
Battle Chess CD32
Battle Toad CD32
Beneath a Steel Sky CD32
Benefactor CD32: Game slow down and blink if you push blue button
Black Viper CD32
Brutal Football CD32: can't select menu
Brutal Paws of Fury CD32
BubbaNStix CD32
Bubble & Squeak CD32

rkrajnc, 2016-11-28:
Hi,

the CD32 gamepad support is not finished (I mentioned that in the release notes). It *should* work, if the docs I have are correct, but unfortunately it doesn't. Hopefully, I'll be able to fix it in some of the upcoming releases.

SheldonRetrogaming, 2016-11-28:
I Thanks for your answer and details.
I will be ready to test : my test list has 93 games.

rkrajnc, 2016-11-30:
Just to be sure, you are using a 'real' CD32 pad connected to the DB9 port, right? This should work *without* enabling the CD32Pad option in the OSD menu, that is intended just for emulating CD32pad for USB devices.

Did you try it without enabling that option? Unfortunately, I don't have any real CD32 pads, so I can't test this myself, but it should *just work*, as there is no special hardware needed to handle this.

SheldonRetrogaming, 2016-11-30:
Just to be sure, you are using a 'real' CD32 pad connected to the DB9 port, right? Yes

Did you try it without enabling that option?  yes, but it does'nt work with CD32 WHDLOAD GAMES
only 2 or 1 button works.

With game  Base Jumper CD32 WHDLOAD + CD32PAD option false or true + usb PAD + CD32PAD option in the game => doesn't work, only 2 buttons work but not left, not right, not up, and not down.

With game Base Jumper CD32 WHDLOAD + CD32PAD option false or true + db9 real CD32PAD + CD32PAD option in the game => doesn't work, only 2 buttons work (Red and Blue) but not left, not right, not up, and not down.

SheldonRetrogaming, 2017-05-29:
[image]
Hello,
have you use the attached files in order to implement cd32 joypad ?

Best Regards.