rkrajnc/minimig-mist #66: Graphic Issue (WHDLOAD)

KristofC, 2016-11-29:
Following games tested with graphic issue.

- Hybris
- Rtype 2

The games were tried on an A1200 AGA system with installed WB3.1 and WHDLOAD.

rkrajnc, 2016-11-30:
Hi, thanks for the report. I'll take a look at it as soon as I can.

AmigaIstMiST, 2016-12-03:
Are you using the "old" or the "new" 1.2 version? I had a graphic issue with Hybris and the "old" version of the 1.2 core. However the "new" version works fine with Hybris for me (plain ADF, no WHDLOAD).

KristofC, 2016-12-03:
Hi! I dont know. I think it is the latest version, downloaded from your Website. I will try again both Version.

AmigaIstMiST, 2016-12-03:
Chaos' website, not mine :-)

rkrajnc, 2017-01-14:
Any update testing these games with the latest release?

KristofC, 2017-01-14:
You mean Version 1.2.1 beta? I just tested Hybris. With the same issue. Now i will test the other games.

KristofC, 2017-01-14:
Hybris - same issue
Virtual Karting 2 - same issue
Rtype 2 - same issue
Fears - it works now fine!

Tested on the same system as before just with the last firmware on 1.2.1 beta.

AmigaIstMiST, 2017-01-14:
ChipRAM turbo option enabled? Should be disabled.

KristofC, 2017-01-14:
Same result with both option.

rkrajnc, 2017-01-15:
Just tried Hybris WHDLoad, and I saw no problems. Is is a specific part of the game, or the whole game?

KristofC, 2017-02-09:
Now i tested the last Beta (V1.2.2) with the last firmware.

Hybris works now correct.
Fears works also correct.
Virtual Karting 2 same issue with wrong colors
Rtype 2 same issue with orange colored stripes on the screen