rkrajnc/minimig-mist #72: Code cleanup/strange constructions.

sorgelig, 2017-01-15:
Hi,
1) While working on YPbPr patch i found compiler generates a lot of blue warnings which makes development harder since it's easier to miss some real problem buried in other non-critical warning. What you think if i will start to work on removing these warnings by removing unused variables or non-used assignments adjustment of widths according to compiler?
2) Why code uses a lot of "<= #1" assignments? Such delay is not synthesized as far as i know.

rkrajnc, 2017-01-15:
Hi,

1. Yes, I'm aware a little cleanup is long overdue ;) I'm just so used to all the warnings that they don't bother me that much. I guess I need to spend some time cleaning the code up.

2. The delay spec for registers is ignored when synthesizing the RTL, so it doesn't make any difference if it's there or not. But for simulation, it is taken into account, and used for a very useful purpose of showing that registers don't change their value at the clock edge, but sometime after that. It makes the simulation waveforms a lot easier to read, plus it avoids a potential simulation error, where the simulator got confused with which signals changed their state in the previous clock cycle (I don't know of any simulator that exhibits this problem atm, but there were such problems some time ago). Ideally, you'd want *all* the registers to have this kind of delay added, except of course anything that generates the clock for the simulation.

sorgelig, 2017-01-16:
ok then.

closed 2017-01-16 by sorgelig