rkrajnc/minimig-mist #73: Sound issue: during intro of "Switch Blade II" game

sblendorio, 2017-01-17:
Please listen to this segment, first 20 secs (actually 1:45 - 2:04 of the video)
https://www.youtube.com/watch?v=aKQlCrlGiuA#t=1m45s

This is the original way the intro sounds like.

Now listen to it played through MiST: the "vibrato" effect disappears completely and the notes played sound quite different.

rkrajnc, 2017-01-27:
Hi,

thanks for the report. This is probably due to minimig missing Amiga's sound modulation capabilities, where one (or multiple) of the channels can AM/FM modulate other channels.

I'll look into implementing this.

Higgy69, 2018-03-06:
Hi. 
Just to let you know I came across someone who has implemented Channel Modulation in Minimig.

It may help you save some effort/work.

I mentioned this on the MiST forum ( Atari-Forum)  and asked if someone could build a new Core with this new Audio Engine added. Thinking about it, I should have mentioned it first to you as your work is the main goto for Amiga Minimig for MiST owners ;)

Here is the info:
http://www.atari-forum.com/viewtopic.php?f=115&t=33324