rkrajnc/minimig-mist #74: Some idea for better Video.

sorgelig, 2017-01-19:
Hi, Rok

as you are more familiar with Amiga internals, i want to ask you if it possible to achieve:

As we know, higher resolutions on standard AGA chipset are tricky and impossible to display for most TVs/Monitors due to very specific hsync/vsync.
If we go for RTG, then it will require some additional knowledge of RTG HW. And may be current MiST SDRAM bandwidth limitation won't allow to make it.

Then I've thought, what if Minimig core will detect some of high AGA resolutions like HD720, HighGFX and will change parameters to make it compatible with standard VESA/HD timings? Since it's mostly for Workbench, some video features like copper/sprites (except mouse pointer) can be turned off while one of these video modes are activated. Precise timings aren't used in WB as well.
So, basically we can have virtual VESA/HD compatible mode with standard AGA chipset without going to RTG. 

Do you think it's possible?

rkrajnc, 2017-01-22:
Hi,

the Amiga video system is locked to custom chipset frequencies, so you are more or less limited in what you can do regarding video output. What the programmable display modes can do is all you can really do, without changing the custom chipset into something that is no longer Amiga, and won't be supported with existing drivers/OS. The best way is to get RTG working, since then you can use pixel frequencies that are not tied to the chipset clock.

You can't really disable copper and blitter and expect workbench to work, since it relies heavily on those two features.

Another possibility would be to re-time the output to better fit some standard VGA/HDMI mode, that would at most require another line buffer. If someone comes up with any such mode (but with the same refresh rate!), I could look into it, but as it is, the really high graphics modes are too slow to be usable anyway, so I'd rather put more work towards RTG support.

sorgelig, 2017-01-23:
If you think, RTG is easier to implement than custom tweaks of AGA chipset, then no problem. Do you have any progress in it?
As i've wrote above, RTG card has it's own on-board memory with frame buffer. Has minimig core any reserved SDRAM time bandwidth to emulate separate memory?

rkrajnc, 2017-01-27:
No, not much progress with RTG.

Minimig on the MiST board has more bandwidth than is needed by the custom chipset, which is currently used by the CPU. This extra bandwidth could be shared with the RTG DMAs, but that would of course mean less memory bandwidth for the CPU, hopefully not so little as to make the CPU much too slow. Something like 640x480x8BPP@60Hz or maybe 800x600x8BPP@60Hz should be doable, though. Theoretically, the Amiga custom chipset bitplane & sprite fetches could be disabled, saving some bandwidth. Well, not really disabled, because of some dependencies on writes happening to the bitplane data registers, but could be changed to a constant, for example.

sorgelig, 2017-01-27:
I think, at least 1024x768@60 is needed (1280x720 would be much better for modern TV/Monitors).
640x480 is even less than standard PAL resolution. 800x600 is also almost like PAL. It's progressive and 256 colors, but for different level of Amiga experience higher resolution is required, i think.

rkrajnc, 2017-01-27:
Well, I don't think such high resolutions will be possible on the MiST board, since the 16bit SDRAM just can't provide the needed bandwidth - a 1024x768 resolution has a minimum pixel clock of 65MHz, that's half of the available SDRAM bandwidth (for a 16bit mode), and I just can't see that working with the current design, since the SDRAM controller has a real-time priority channel for the amiga custom chipset, and you can't really expect to get full SDRAM bandwidth out of it.

kowoba, 2017-01-27:
Maybe time for a new and improved MIST :)