rkrajnc/minimig-mist #75: Support for PCMCIA storage

sorgelig, 2017-01-23:
It would be good if Minimig will be able to use PCMCIA virtual CF card. I think most modules used for IDE can be adapted for PCMCIA. It should be just another IDE port (with small variation), isn't?

PCMCIA storage support would be very useful! It will allow to mount primary partition of SD card as a CF with FAT95 driver and do import/export inside Minimig. Currently i'm using second partition of SD card as a HDD for Minimig, and i cannot access it easily in Windows. Thus using primary SD card's partition as CF will allow me to transfer files within Minimig.

rkrajnc, 2017-01-27:
Hi,

while it would certainly be nice to use more than two HD partitions/images, what you require can already be done with the existing HD support. You can mount the SDcard's FAT partition as the second HD drive and use it on AmigaOS. (BTW, you can also mount a CDROM .iso image as a hard disk in minimig),

I will look into adding PCMCIA hard disk support, but it will have to wait for more pressing issues to be resolved first.

sorgelig, 2017-01-27:
I know i can mount FAT partition as HDD, but i never saw any guide how to setup FAT partition on Amiga as a standard IDE HDD. I only know how to setup PCMCIA storage for FAT.
And of course if you want to transfer file to a non-bootable Amiga HDD, you cannot do this if you only can mount 2 HDDs and they both are non-bootable. Of course you can boot from FDD or do double transfer. But in this case it would be easier to use WinUAE then.

rkrajnc, 2017-01-27:
Well, you mount the .hdf file, which can be bootable, as the first disk, and the whole SD card FAT partition as the second disk. This way, you can transfer files from the SD card FAT partition to the .hdf disk image. Am I missing something?

sorgelig, 2017-01-27:
Yeah.. Missing the instruction how to setup FAT driver for HDD. Probably, even more complicated: how to setup 2 file systems for one HDD. Because if i will swap second HDD between FAT partition and SFS partition, AmigaOS should recognize both types of partitions. My AmigaOS has fat95 driver since it's image from real Amiga where fat95 works fine with PCMCIA-CF storage. If i mount FAT partition as HDD in minimig it's not recognized. Actually HDD requires RDB boot block which is obviously missing if i mount FAT partition.