rkrajnc/minimig-mist #77: USB mouse not recognized

sblendorio, 2017-02-11:
Hi, I have a nice USB mouse that is not recognized my MinimigMist. What kind of information can I report to try understanding why?

sblendorio, 2017-02-12:
Those are device-related USB info I retrieved from Windows... Hope it could help finding infos...

[image]

rkrajnc, 2017-02-13:
Hi,

did you try any other cores, does your mouse work there? I presume it doesn't, since the ARM controller handles USB devices, and it probably has problems detecting your mouse.

I would have to see the debug output from the ARM microcontroller. There is a guide how to do that here:
https://github.com/mist-devel/mist-board/wiki/Troubleshooting#usb
You will have to use the AtariST core for this. Connect your MiST board to the PC with the USB (power) cable, start the AtariST core, enable USB debugging, and make sure you see something on your serial terminal. Then connect your mouse and watch the output on the serial monitor. Please attach the output here.

Another possibility is to enable mouse boot mode  in your mist.ini and see if that helps:
https://github.com/mist-devel/mist-board/wiki/DocIni#mouse_boot_mode

sblendorio, 2017-02-13:
Hi rkrajnc, thanks for your answer. First I tried to set "mouse_boot_mode=1" in "mist.ini", but I got no change.

Second, I got the following output from USB debug: I started AtariST with no mouse connected, then I turned on Debug on USB, then I connected, then disconnected the mouse

    key 0x28 break: 0x1C
    key 0x29 make: 0x1
    key 0x29 break: 0x1
    key 0x29 make: 0x1
    key 0x29 break: 0x1
    OSD is now invisible
    status change on port 1, 0x10100
    Status of port 1:
     powered
    Changes on port 1:
     connected
     port 1 disconnect!
    usb_release_device(parent=1, port=1
    status change on port 1, 0x10301
    Status of port 1:
     connected
     powered
     low speed
    Changes on port 1:
     connected
     dev 1: port 1 connect!
    resetting port 1
    status change on port 1, 0x100303
    Status of port 1:
     connected
     enabled
     powered
     low speed
    Changes on port 1:
     reset
     port 1 reset complete!
    usb_configure(parent=1 port=1 lowspeed=1)
    using free entry at 2
    Setting addr 3
    usb_set_addr(new=3)
    failed to assign address
    status change on port 1, 0x10100
    Status of port 1:
     powered
    Changes on port 1:
     connected
     port 1 disconnect!
    usb_release_device(parent=1, port=1