rkrajnc/minimig-mist #79: Nexus 7 demo doesn't work and other issues

vebxenon, 2017-04-10:
Since minimig_mist-20160224 and new betas, Nexus 7 demo starts, but it remains always showing the Andromeda animation once the music changes, so no effects and nothing are shown in this demo.

Also, I tested yesterday Time Gal http://unofficial-cd32-ports.blogspot.com.es/2017/03/time-gal-reimagine.html and the game hangs when you reach the 4th level and sometimes in demo mode, both in last stable release and last beta.

Finallly, iGame https://github.com/MrZammler/iGame/tree/master/iGame_rel https://github.com/MrZammler/iGame only works with the "NOSCREENSHOT" option, and even copying https://github.com/MrZammler/iGame/blob/master/guigfx_render_nofpu.lha the NOFPU libs, it shows 8000000B guru error when starting if I want to see game screenshots.

Regards,

Salva

rkrajnc, 2017-04-10:
Hi,

thanks for the report. I'm pretty sure I had WHDload'ed Nexus7 running just fine with one of the latest builds. It's probably sensitive to turbo setting (so make sure you have AGA, 68020, some fastRAM enabled and try to toggle turbo option).

I'll try Time Gal (never heard of it before, thanks!).

iGame seems to be problematic, but at least one version (from ClassicWB) seems to work just fine, so this is low priority for me (especially as it is 99.9% caused by the CPU softcore).

Just a side note: next time please open a separate issue for each bug you report, it just makes it easier to manage.

solskogen, 2018-11-04:
The demo seems to work fine with latest beta.

apolkosnik, 2019-07-13:
There's a graphical glitch in the Goraud pulse part, I think that BFSET instruction implementation in tg68  might be to blame.