rkrajnc/minimig-mist #82: USB joystick disables DB9 mouse

AmigaIstMiST, 2017-09-23:
- plugin a real DB9 Amiga ("tank") mouse

Result: the mouse works, everything is fine

- plugin a USB joystick/joypad additionally to the DB9 Amiga mouse

Result: the USB joystick disables the DB9 Amiga mouse!
If you unplug the USB joystick, the DB9 Amiga mouse works again.

AmigaIstMiST, 2017-09-23:
Tested with firmware ATH170207 and MinimigAGA cor 1.2.2 BETA.

rkrajnc, 2017-09-23:
Hi,

yes there is some code in the firmware that reorders joysticks / mice when an USB device is plugged in. It's annoying, I agree. I'll take a look at it and try to come up with a solution.

rkrajnc, 2017-09-24:
Hi,

try to update your MiST firmware with the attached file (unzip it to the root of your SD card), and add an entry to your mist.ini file, under the [mist] section:
`joystick0_prefer_db9=1`

I only tried it with a joystick in JOY1 and AAA, but it should work with a mouse also. Please report if it works for you so I can close the issue and push the code.

[firmware.zip](https://github.com/rkrajnc/minimig-mist/files/1327648/firmware.zip)

AmigaIstMiST, 2017-09-24:
Man, thats fast! 😊
I will test it tomorrow evening.

AmigaIstMiST, 2017-09-25:
Ok, I tested all 4 possible combinations of USB/DB9 mice and joysticks:

- usb mouse, usb joypad
- usb mouse, db9 joystick
- db9 mouse, usb joypad
- db9 mouse, db9 joystick

Everything works as expected!

However, it is a little odd, that an extra INI entry is needed for this. I think, this this the expected behavior!

rkrajnc, 2017-09-25:
Hi,

thanks for testing this. Actually, the extra ini setting is really needed, as the firmware can't automatically detect DB9 - connected devices, so if a USB joystick is present, it has to assume that no DB9 joysticks are present and make the USB device the primary joystick (so at least one device works).

That is how the firmware works without the new ini entry - as soon as one USB device is present, it makes it joystick number two, and makes DB9 JOY1 the third device. The new ini entry disables this joystick reordering.

Theoretically, the firmware could detect the DB9 - connected mouse or joystick when it would detect direction movement or a button press and then reorder the devices, but that would make it even more confusing, so for now, this solution should do.

rkrajnc, 2017-09-25:
Fixed in MiST firmware with commit 23ad06e0fdaa629092f97c9aca62b32a522bf6ff.

closed 2017-09-25 by rkrajnc

AmigaIstMiST, 2017-09-25:
Can the firmware detect the difference between a USB mouse and a USB joystick?
If yes, what about this simple logic:

Is a USB mouse present? If no, assume a DB9 mouse in the first DB9 port.

AmigaIstMiST, 2017-09-25:
Hmm, but this could cause problems with the C64 core, where normally no mouse is attached at all, so forget my comment. At least it is possible now to use the Amiga mouse as main mouse. As it is possible now to turn your old Amiga mouse into a laser mouse with these new laser upgrade kits, this is becoming more important now.

AmigaIstMiST, 2017-10-31:
What about releasing the firmware as a binary ? :-)

rkrajnc, 2017-11-03:
What do you mean? The binary is 5 posts up, the [firmware.zip](https://github.com/rkrajnc/minimig-mist/files/1327648/firmware.zip) link.

AmigaIstMiST, 2017-11-03:
I mean: making it public in the official MiST repo. You did a code check
in, but you did not check in the binary.