rkrajnc/minimig-mist #84: Cache clear flag possible issue

sorgelig, 2018-04-24:
https://github.com/rkrajnc/minimig-mist/blob/da69f6ec45f9956f1df56f0a9a7f4a062c9b726e/rtl/sdram/cpu_cache_new.v#L214

CACR[3] is write-only bit, so CPU cannot read the last state, thus if it will write 1 second and more times it won't trigger the clear since internally it's 1 already. Isn't?

rkrajnc, 2018-04-24:
I didn't implement this correctly in the TG68 code - the C & CE bits should not be latched, it should be just a pulse that should signal the cache to clear itself.

sorgelig, 2018-04-25:
Ok. At least i understood it correctly :)
I'm trying to replace TG68 core with 68EC30 core i've found. Unfortunately Minimig uses some specific to TG68 signals making hard to port standard CPU core.

rkrajnc, 2018-04-25:
Can you share the 030 core, I can take a look at it and see if anything can be done quickly?

sorgelig, 2018-04-25:
Sure! Would be good if you will help. 
It's from Suska project: https://download.experiment-s.de/Configware/rtl/vhdl.zip
Inside you can find 68EC30: vhdl\68K30L

sorgelig, 2018-04-25:
You will need to add 32bit->16bit bus glue logic which you can get from here: https://github.com/terriblefire/tf530 
It has RTL code, so pretty much can be used as-is in minimig.

apolkosnik, 2019-07-13:
https://download.experiment-s.de/Configware/2K19A/rtl/vhdl.zip